Stachugit/CatHack #6: SubSend/SubRecord not working?

bisaw37, 2024-09-04:
For whatever reason, whenever I record a signal, It registers and it looks to function properly, but If I try to send it, nothing happens. At all. Its like it thinks it sends the signal, but it never does. I've made sure all my wires are soldered correctly, and jamming seems to work fine, so it can send info, but it just never seems to work with recording.

Stachugit, 2024-09-04:
> For whatever reason, whenever I record a signal, It registers and it looks to function properly, but If I try to send it, nothing happens. At all. Its like it thinks it sends the signal, but it [...]

it depends on the signal, it is possible that the signal has too wide a bandwidth and therefore cannot be recorded correctly, what are you trying to copy?

bisaw37, 2024-09-04:
Im trying to record a signal from a cheap smart plug’s remote. The analyze function says it’s 433.92 mhz. Also If I go into monitor, I can see the waveform, so it is absolutely receiving the signal.

Stachugit, 2024-09-06:
> Im trying to record a signal from a cheap smart plug’s remote. The analyze function says it’s 433.92 mhz. Also If I go into monitor, I can see the waveform, so it is absolutely receiving the signal.

Please check my post on GitHub, it should fix your issue:

https://www.reddit.com/r/M5Stack/s/CUjuQbeYnU

closed 2024-09-06 by Stachugit

alex-btc, 2024-09-28:
I also have this problem.

Recording RF (433MHz) signals is ok. They are saved to the SD card without any problem.

When trying to transmit them, it does nothing. I analysed the spectrum with other stickCplus2 and really when trasmiting with the unit numer 1, nothing is captured by unit number 2 (monitoring spectrum).

All setup, wiring is all right.