Stachugit/CatHack #9: Faulty CC1101 module?

turbochargedkid, 2024-09-21:
Hi, I've rewired this a few times and am still just getting constant static in the subghz monitor screen. Has anybody else managed to fix this or could I just have a faulty CC1101 module?
[image]

peppemas1, 2024-09-27:
i have same problem, i build the module cc1101 + sd card  and work fine , after some day  i try subghz monitor and get this strange wave like the photo you post. After i change the cc1101 thinking is broken, after resoldering the new module nothing , same problem, i recheck connection many times with multimeter , secure 100% correct. I try a rfid2 mstack original module with bruce and i get error module rfid not connected, i think sda and slc (G32 and G33) grove port are damaged, i order a new m5stickc , i still waiting to arrive

peppemas1, 2024-09-27:
check on google in a japanese site i translate and i found the the problem are the 2 diode near the grove port
[image]

Stachugit, 2024-09-28:
it is possible that the pins simply shorted and broke

peppemas1, 2024-09-28:
no pins are good, maybe possible i have connected g32 and g33 inverse, in the new module now i use an original grove connector , i still waiting to arrive the new m5stick for try

turbochargedkid, 2024-10-03:
Ah that's a shame that it's likely damage then. Thanks for the updates 🙂

peppemas1, 2024-10-03:
possible also m5stickc defective , not damaged with short
the test with my cc1101 module
https://www.instagram.com/reel/DAoaLyONcmr/?igsh=ZnlhZGZoN2IwNGhy

hi Stachu , we waiting the new firmware version with rfid support