Stachugit/CatHack #13: SubSend RF signals not working

alex-btc, 2024-09-29:
Recording RF (433MHz) signals is okay. They are saved to the SD card without any problem.

When trying to transmit the signals, it does nothing. I analyzed the spectrum with other stickCplus2 and when transmitting with the unit number 1, nothing is captured by unit number 2 (monitoring spectrum).

All setup, wiring is right.

peppemas1, 2024-10-03:
try monitor spectrum with m5 number1 , if work correctly. In my m5stickc grove port damaged after some days

alex-btc, 2024-10-03:
I tried send signals with another stickCplus2, to check if the first one was damaged, but the problem persisted.

Signal saving is ok, but signal transmitting no. The screen says "Signal sent", but nothing shows up in the other signal monitor.

I even tested it with another CC1101 module, but the problem still persisted.