Stachugit/CatHack #18: Guys, please create a firmware for the ESP32-2432S028 someday.

Asmodeux2, 2024-11-11:
This platform is very common, and I would like to see your firmware on this device. Besides your firmware on ESP32-2432S028, there are Marauder, Bruce, Ghost_ESP and etc.

[image]

Trhug25, 2024-11-15:
What are the base programs that run on cyd?

Asmodeux2, 2024-11-15:
> What are the base programs that run on cyd? 

The basic ones are WiFi and Bluetooth. As on all ESP32s. It is also possible to connect modules via GPIO.

Asmodeux2, 2024-11-21:
@Trhug25 https://github.com/radio3-network/kit-ESP32-2432S028R/tree/main/downloads/2.8inch_ESP32-2432S028R Base programms and Documentation